proveskit/fprime-proves #5: Add Continuous Integration

Lex-ari, 2024-08-14:
It should mirror some of the checks listed here: https://github.com/nasa/fprime/tree/devel/.github/workflows
While this one is mainly testing fprime itself, it has files for testing a full build pipeline.
See: https://github.com/nasa/fprime/blob/devel/.github/workflows/ext-build-hello-world.yml

This is a copy of SCALES' [CI Issue](https://github.com/BroncoSpace-Lab/scales-software/issues/3) and [PR](https://github.com/BroncoSpace-Lab/scales-software/pull/4). CI for both are probably nearly identical, and work should be carried over if it is the case.

nateinaction, 2024-08-15:
I've reviewed the workflows in the fprime repo and I think there's a lot to learn from in terms of their static analysis and testing setup. I'm going to try to use the [composite action](https://github.com/fprime-community/project-builder/blob/main/action.yml) in the community builder repo. I'm choosing the composite action over the reusable workflow because we need to install some additional tooling like the arduino CLI and board downloads before the build.

nateinaction, 2024-08-20:
Completed! 🚀

closed 2024-08-20 by nateinaction