proveskit/fprime-proves #9: Rebase mainline branch to make `.git` directory smaller

nateinaction, 2024-08-16:
Hey team,

Due to the previous inclusion of the python venv files removed in #2, we have a `.git` directory that's 175MB large. This is probably fine for most folks with a high speed connection but we could improve developer experience during the cloning/forking process by rewriting the history for the repo's first commit.

Rewriting history in git can be a big deal if we have many changes and contributors. However, if we believe this repository will be here for the long term, it feels like a good thing to do now while its contributor list is small.

From this repo:
```sh
du -h -d 1 ./.git/objects/pack
175M	./.git/objects/pack
```

Example of another fprime repo:
```
du -h -d 1 ./.git/objects/pack
  0B	./.git/objects/pack
```

Thoughts?

Mikefly123, 2024-08-18:
This change looks good to me! I went ahead and made you a repo admin so you can do what needs to be done.

nateinaction, 2024-08-20:
Done! 🙃 
```sh
du -h -d 1 ./.git/objects/pack
612K	./.git/objects/pack
```

closed 2024-08-20 by nateinaction