proveskit/fprime-proves #14: Update ReadMe

Lex-ari, 2024-08-23:
More details to add to the readme:
```shell 
sudo chmod 777 /dev/ttyACM0
```

nateinaction, 2024-08-25:
Hmmm, this sounds like something specific on your machine. I don't think I would recommend `chmod 777 /dev/ttyACM0` unless folks understand the security implications of doing that. Any user with the right groups should be able to start a serial connection without 777 perms on that device.

Lex-ari, 2024-08-25:
I do admit it may be specific to linux environments, and was tested with a fresh install. Fprime-gds may not have access rights to things in /dev, which is why I've done this fix, (and I think running `sudo fprime-gds` is discouraged).