proveskit/fprime-proves #49: Weird Mac Issue?

Mikefly123, 2024-11-05:
[image]

Lex-ari, 2024-11-05:
the problem is is that you're not on linux

nateinaction, 2024-11-05:
Not another submodule issue! 😄 

I haven't been having issues with this and I've been switching between branches with a lot of work-in-progress. Many times I find that a submodule exists in one branch but not in another. To make sure I can build I run `git submodule update --init --recursive ./SUBMODULE_DIR` after I check out a change.

Mikefly123, 2024-11-05:
@nateinaction with a bit of wrenching this might be an upstream GitHub Desktop issue? When I use the command line everything is fine. Perhaps the way we are setting up the submodules is not agreeing with the way GitHub Desktop wants to pull things in?

closed 2025-08-25 by Mikefly123